NVIDIA-ISAAC-ROS/isaac_ros_visual_slam #140: Package Location Error—— “E: Unable to locate package ros-humble-isaac-ros-visual-slam”

hjc0106, 2023-12-29:
When executing `./scripts/run_dev.sh ${ISAAC_ROS_WS}`  to launch the docker container, it failed due to network connection——"Err:31 http://ppa.launchpad.net/mosquitto-dev/mosquitto-ppa/ubuntu focal InRelease                                                                                        Cannot initiate the connection to ppa.launchpad.net:80 (2620:2d:4000:1::81). - connect (101: Network is unreachable) Could not connect to ppa.launchpad.net:80 (185.125.190.80), connection timed out". Perhaps it won't affect the installation of visual SLAM?
Then I tried to `sudo apt-get install -y ros-humble-isaac-ros-visual-slam`, package location error occurred——“E: Unable to locate package ros-humble-isaac-ros-visual-slam”. I am running on Ubuntu 22.04.

The full log is provided below:
Run `./scripts/run_dev.sh ${ISAAC_ROS_WS}`

Running isaac_ros_dev-x86_64-container
Get:1 https://repo.download.nvidia.cn/jetson/x86_64/focal r35.4 InRelease [3,302 B]                                                                                         
Get:2 http://packages.ros.org/ros2-testing/ubuntu focal InRelease [4,685 B]                                                                                                 
Get:4 http://archive.ubuntu.com/ubuntu focal InRelease [265 kB]                                                                                                             
Get:5 https://repo.download.nvidia.cn/jetson/x86_64/focal r35.4/main amd64 Packages [26.8 kB]                                                                               
Get:6 http://security.ubuntu.com/ubuntu focal-security InRelease [114 kB]                                                                                                   
Get:8 https://apt.kitware.com/ubuntu focal InRelease [15.5 kB]                                                                                                              
Get:7 https://isaac.download.nvidia.cn/isaac-ros/ubuntu/main focal InRelease [4,316 B]                                                                                      
Get:9 https://isaac.download.nvidia.cn/isaac-ros/ubuntu/main focal/main amd64 Packages [473 kB]                                                                             
Get:10 https://developer.download.nvidia.cn/compute/cuda/repos/ubuntu2004/x86_64  InRelease [1,581 B]                                                                       
Get:11 https://developer.download.nvidia.cn/compute/cuda/repos/ubuntu2004/x86_64  Packages [1,317 kB]                                                                       
Get:12 https://apt.kitware.com/ubuntu focal/main amd64 Packages [74.7 kB]                                                                                                   
Get:3 https://packagecloud.io/github/git-lfs/ubuntu focal InRelease [28.0 kB]                                                                                               
Get:13 http://security.ubuntu.com/ubuntu focal-security/universe amd64 Packages [1,149 kB]                                                                                  
Get:14 http://archive.ubuntu.com/ubuntu focal-updates InRelease [114 kB]                                                                                                 
Get:15 http://archive.ubuntu.com/ubuntu focal-backports InRelease [108 kB]                                                                                                  
Get:16 http://archive.ubuntu.com/ubuntu focal/multiverse amd64 Packages [177 kB]                                                                                            
Get:18 http://archive.ubuntu.com/ubuntu focal/restricted amd64 Packages [33.4 kB]                                                                                           
Get:19 http://archive.ubuntu.com/ubuntu focal/main amd64 Packages [1,275 kB]                                                                                                
Get:17 https://packagecloud.io/github/git-lfs/ubuntu focal/main amd64 Packages [3,505 B]                                                                                    
Get:20 http://security.ubuntu.com/ubuntu focal-security/main amd64 Packages [3,283 kB]                                                                                      
Get:21 http://archive.ubuntu.com/ubuntu focal/universe amd64 Packages [11.3 MB]                                                                                             
Get:22 http://security.ubuntu.com/ubuntu focal-security/multiverse amd64 Packages [29.3 kB]                                                                                 
Get:23 http://security.ubuntu.com/ubuntu focal-security/restricted amd64 Packages [3,130 kB]                                                                                
Get:24 http://archive.ubuntu.com/ubuntu focal-updates/main amd64 Packages [3,761 kB]                                                                                        
Get:25 http://archive.ubuntu.com/ubuntu focal-updates/restricted amd64 Packages [3,279 kB]                                               
Get:26 http://packages.ros.org/ros2-testing/ubuntu focal/main amd64 Packages [1,191 kB]                                                  
Get:27 http://archive.ubuntu.com/ubuntu focal-updates/multiverse amd64 Packages [32.0 kB]                                                              
Get:28 http://archive.ubuntu.com/ubuntu focal-updates/universe amd64 Packages [1,444 kB]                                                             
Get:29 http://archive.ubuntu.com/ubuntu focal-backports/universe amd64 Packages [28.6 kB]                                                              
Get:30 http://archive.ubuntu.com/ubuntu focal-backports/main amd64 Packages [55.2 kB]                                                                
Err:31 http://ppa.launchpad.net/mosquitto-dev/mosquitto-ppa/ubuntu focal InRelease                                                                                          
  Cannot initiate the connection to ppa.launchpad.net:80 (2620:2d:4000:1::81). - connect (101: Network is unreachable) Could not connect to ppa.launchpad.net:80 (185.125.190.80), connection timed out
Fetched 32.8 MB in 2min 44s (200 kB/s)                                                                                                                                      
Reading package lists... Done
W: Failed to fetch http://ppa.launchpad.net/mosquitto-dev/mosquitto-ppa/ubuntu/dists/focal/InRelease  Cannot initiate the connection to ppa.launchpad.net:80 (2620:2d:4000:1::81). - connect (101: Network is unreachable) Could not connect to ppa.launchpad.net:80 (185.125.190.80), connection timed out
W: Some index files failed to download. They have been ignored, or old ones used instead.
reading in sources list data from /etc/ros/rosdep/sources.list.d
Hit file:///etc/ros/rosdep/sources.list.d/nvidia-isaac.yaml
Hit https://raw.githubusercontent.com/ros/rosdistro/master/rosdep/osx-homebrew.yaml
Hit https://raw.githubusercontent.com/ros/rosdistro/master/rosdep/base.yaml
Hit https://raw.githubusercontent.com/ros/rosdistro/master/rosdep/python.yaml
Hit https://raw.githubusercontent.com/ros/rosdistro/master/rosdep/ruby.yaml
Hit https://raw.githubusercontent.com/ros/rosdistro/master/releases/fuerte.yaml
Query rosdistro index https://raw.githubusercontent.com/ros/rosdistro/master/index-v4.yaml
Skip end-of-life distro "ardent"
Skip end-of-life distro "bouncy"
Skip end-of-life distro "crystal"
Skip end-of-life distro "dashing"
Skip end-of-life distro "eloquent"
Skip end-of-life distro "foxy"
Skip end-of-life distro "galactic"
Skip end-of-life distro "groovy"
Add distro "humble"
Skip end-of-life distro "hydro"
Skip end-of-life distro "indigo"
Add distro "iron"
Skip end-of-life distro "jade"
Skip end-of-life distro "kinetic"
Skip end-of-life distro "lunar"
Skip end-of-life distro "melodic"
Add distro "noetic"
Add distro "rolling"
updated cache in /home/admin/.ros/rosdep/sources.cache
 * Stopping hotplug events dispatcher systemd-udevd                                                                                                                   [ OK ] 
 * Starting hotplug events dispatcher systemd-udevd                                                                                                                   [ OK ] 

Run `sudo apt-get install -y ros-humble-isaac-ros-visual-slam`

Reading package lists... Done
Building dependency tree       
Reading state information... Done
E: Unable to locate package ros-humble-isaac-ros-visual-slam

pgaston, 2023-12-30:
Seems like this is a recent problem?   See as well  --  https://forums.developer.nvidia.com/t/run-dev-sh-failed-to-build-base-image-file-has-unexpected-size/277156/6

Bharannn, 2024-02-16:
Hi @hjc0106 , I am facing the same issue, were you able to solve this issue or did you find any workaround and also I am running setup.bash each time I run docker, I am trying to add the command to bashrc but each time the a new container is created the old bashrc is getting removed. Please let me know if you can help me with these things.

Thank you.

swapnesh-wani-nvidia, 2024-02-21:
Hello, could you share the full logs here? This looks like an intermittent failure.