NVIDIA-ISAAC-ROS/isaac_ros_visual_slam #141: Unable to use VisualSlamNode with intra-process comms when running with composition

tanelikor, 2024-01-04:
When running the VisualSlamNode with as composed in the same component container with e.g. the camera driver, it would be desirable to enable intra-process comms to enable transmitting the camera data more efficiently. However, this is not possible in the latest version v2.1.0, because there is apparently some topic with a transient local QoS included in the intra-process topics. Previously in 0.31.0-dp (which we just upgraded from) using intra-process comms worked.

I did not have time to thoroughly investigate which topic causes this, but I suspect the publisher to `/tf_static` (I did check most of the topics and everything that I saw had a volatile durability, while `/tf_static` obviously will have the transient local one). I am not sure why this node would need to publish static transforms, but if it is needed, it would be good to exclude it from the intra-process stuff.

For reference, here is how my launch of the node essentially looks like (this is part of a quite large launch file, so I'm leaving out some other nodes etc that are irrelevant for this issue):

```
def generate_launch_description():
    visual_slam_node = ComposableNode(
        name="visual_slam_node",
        namespace="camera_1",
        package="isaac_ros_visual_slam",
        plugin="nvidia::isaac_ros::visual_slam::VisualSlamNode",
        extra_arguments=[{"use_intra_process_comms": True}],  # including this line breaks the launch
        parameters=[
            {
                "denoise_input_images": False,
                "rectified_images": True,
                "enable_imu_fusion": False,
                "enable_debug_mode": False,
                "debug_dump_path": "/tmp/elbrus_1",
                "enable_slam_visualization": False,
                "enable_localization_n_mapping": False,
                "enable_landmarks_view": False,
                "enable_observations_view": False,
                "publish_map_to_odom_tf": False,
                "publish_odom_to_base_tf": False,
                "map_frame": "map",
                "odom_frame": "odom",
                "base_frame": "base_link",
                "input_base_frame": "base_link", 
                "input_left_camera_frame": "camera_1_infra1_frame",
                "input_right_camera_frame": "camera_1_infra2_frame",
                "input_imu_frame": "camera_1_imu_optical_frame",
            }
        ],
    )
    cam_container = ComposableNodeContainer(
        name="camera_1_container",
        namespace="camera_1",
        package="rclcpp_components",
        executable="component_container",  # it does not matter which container you use for this
        emulate_tty=True,
        composable_node_descriptions=[visual_slam_node],
        output={"both": {"screen", "log", "own_log"}},
        arguments=["--ros-args", "--log-level", "info"],
    )
    return LaunchDescription([cam_container])
```

Here are the relevant parts of the logs that I see when launching with the above setup. Note the error about intraprocess communication only being allowed with volatile durability.
```
[INFO] [component_container_isolated-1]: process started with pid [82]
[component_container_isolated-1] [INFO] [1704372041.453758765] [camera_1.camera_1_container]: Load Library: /karelics_workspace/karelics_brain/package_ws/install/isaac_ros_visual_slam/lib/libvisual_slam_node.so
[component_container_isolated-1] [INFO] [1704372041.767271885] [camera_1.camera_1_container]: Found class: rclcpp_components::NodeFactoryTemplate<nvidia::isaac_ros::visual_slam::VisualSlamNode>
[component_container_isolated-1] [INFO] [1704372041.767408176] [camera_1.camera_1_container]: Instantiate class: rclcpp_components::NodeFactoryTemplate<nvidia::isaac_ros::visual_slam::VisualSlamNode>
[ERROR] [launch_ros.actions.load_composable_nodes]: Failed to load node 'visual_slam_node' of type 'nvidia::isaac_ros::visual_slam::VisualSlamNode' in container '/camera_1/camera_1_container': Component constructor threw an exception: intraprocess communication allowed only with volatile durability
[component_container_isolated-1] [ERROR] [1704372043.184402808] [camera_1.camera_1_container]: Component constructor threw an exception: intraprocess communication allowed only with volatile durability
```

hemalshahNV, 2024-01-09:
Yes, you are correct. There is a tf static transform publisher to publish a transform for `map_pose_odom` when enabling "odometry-only" mode using the SetOdometry service where there is no relation between map and odom frames otherwise. We're looking into how we could avoid this or having intraprocess comms option ignore it.

tanelikor, 2024-01-09:
@hemalshahNV Thank you for the response and for confirming my suspicions about the static broadcaster. The fix should then be to simply disable the intra-process comms for the static broadcaster, which should take only a couple lines of code. Would you like for me to make a small PR about this?

Rnfudge02, 2024-12-27:
Is this enhancement still planned?

FahYen, 2025-08-29:
https://forums.developer.nvidia.com/t/intra-process-communication-between-nitros-non-nitros-node/342800

According to Nvidia, you can't avoid zero-copy between Non-Nitros and Nitros node. VisualSlamNode simply can't use intra process communication.