NVIDIA-ISAAC-ROS/isaac_ros_visual_slam #142: RVIZ VIO and SLAM paths seperate

lowellm91, 2024-01-10:
Hi,

I have been using your SLAM software and I have been having some issues. First the VIO and SLAM paths in RVIZ seperate and get extremely far apart. Is this normal? 

Secondly it's not at all clear what features being track actually means, is there is a way to display the features detected ontop of the image pairs so we can diagnose potential problems with tracking and see where tracking fails?

Third the information on frames is pretty confusing, to clarify what is the difference between base_frame and input_base_frame and base_link? My assumption was input_base_frame transforms the base_link to the left_camera_frame but the wording is very confusing.

lowellm91, 2024-01-20:
Messing with this more, changing the input_base_frame to the same name as the left_camera_frame fixed it, I assumed input_base_frame would be the parent name of the transform "left_camera_frame" but apparently this didn't work. It's not clear to what the difference between "left_camera_frame" and "input_base_frame" transforms are now since both transform the base_frame to the left camera.

swapnesh-wani-nvidia, 2024-02-05:
Hello,

>  First the VIO and SLAM paths in RVIZ separate and get extremely far apart. Is this normal?
Ideally, they would stay on top of each other but with any odometry system, you could expect a drift in the VO path as it gives the smooth poses out as opposed to the SLAM path which can have jumps but is accurate w.r.t a `map` frame

>  Secondly it's not at all clear what features being track actually means, is there is a way to display the features detected ontop of the image pairs so we can diagnose potential problems with tracking and see where tracking fails?
You can use the Camera display in RViz and use the observations topic (visual_slam/vis/observations_cloud) as an overlay

>  what is the difference between base_frame and input_base_frame and base_link?
The idea is to add the ability to separate input and output subtrees in tf2.
User can have in_left/in_right/in_base_link and out_map/out_odom/out_base_link.
 For example if you want to compare the 
o/p of Isaac ROS Visual Slam with a ground truth system you can have
in_left/in_right/in_base_link   --- From groundtruth system 
cuvslam_out_map/cuvslam_out_odom/cuvslam_out_base_link   ---- From Isaac ROS Visual Slam

In case of a single robot `input_base_frame` = `base_frame`. If you want to track the motion of the robot put the value of `base_link` in `base_frame`. If you want to track the motion of left imager then put the value of  `left_camera_frame` in `base_frame`.

noelblunder, 2025-04-10:
I think, I have a similar problem with ROS2 Humble and for me, changing the tracking frame from `base_link` to the camera frame did not do the trick. I'm trying to evaluate the package for a relatively large outdoor area, so I know, this might not be the ideal testing area, but the potential would be huge, if the package could be transferred from indoor use cases to localizing in large outdoor scenarios.

As you can see, the SLAM (green) and the VO (blue) paths have a significant offset and are also quite quite far off the ground truth (red), but my main concern is the offset between the two paths. It seems like altering the `slam_max_map_size` and `path_max_size` changes how much the paths are diverging, but nothing seems to scale both paths similarly. 

[image]

here is my launch configuration

```python
    visual_slam_node = ComposableNode(
        name='visual_slam_node',
        package='isaac_ros_visual_slam',
        plugin='nvidia::isaac_ros::visual_slam::VisualSlamNode',
        namespace='mrcd_robot',
        parameters=[{
            'enable_image_denoising': False,
            'rectified_images': False,
            'enable_slam_visualization': True,
            'enable_landmarks_view': True,
            'enable_observations_view': True,
            'camera_optical_frames': ['zed_front_left_camera_optical_frame', 'zed_front_right_camera_optical_frame'],
            'base_frame': 'base_link',
            'num_cameras': 2,
            'enable_imu_fusion': True,
            'imu_frame': 'zed_front_imu_link',
            'gyro_noise_density': 0.004991999827325344,
            'gyro_random_walk': 0.042399998754262924,
            'accel_noise_density': 0.0004400000034365803,
            'accel_random_walk': 0.02019999921321869,
            'calibration_frequency': 200.0,
            'image_jitter_threshold_ms': 68.0,
            'enable_ground_constraint_in_odometry' : False,
            'enable_ground_constraint_in_slam' : False,
            'slam_max_map_size': 10000000,
            'path_max_size': 10000000
        }],
        remappings=[
            ('visual_slam/image_0', 'zed_node/left_raw/image_raw_color_rgb8'),
            ('visual_slam/image_1', 'zed_node/right_raw/image_raw_color_rgb8'),
            ('visual_slam/imu', 'zed_node/imu/data_raw'),
            ('visual_slam/camera_info_0', 'zed_node/left_raw/camera_info'),
            ('visual_slam/camera_info_1', 'zed_node/right_raw/camera_info'),
```
the topics of the bag file, I'm using to test this:

```
Topic information: Topic: /mrcd_robot/ground_truth | Type: nav_msgs/msg/Path | Count: 29872 | Serialization Format: cdr
                   Topic: /mrcd_robot/rs_front/rs_front/depth/image_rect_raw | Type: sensor_msgs/msg/Image | Count: 14942 | Serialization Format: cdr
                   Topic: /mrcd_robot/tf | Type: tf2_msgs/msg/TFMessage | Count: 395975 | Serialization Format: cdr
                   Topic: /mrcd_robot/teleop/joy | Type: sensor_msgs/msg/Joy | Count: 178407 | Serialization Format: cdr
                   Topic: /mrcd_robot/rs_front/rs_front/aligned_depth_to_color/image_raw | Type: sensor_msgs/msg/Image | Count: 14943 | Serialization Format: cdr
                   Topic: /mrcd_robot/velodyne_points | Type: sensor_msgs/msg/PointCloud2 | Count: 9686 | Serialization Format: cdr
                   Topic: /mrcd_robot/zed_node/left_raw/camera_info | Type: sensor_msgs/msg/CameraInfo | Count: 29835 | Serialization Format: cdr
                   Topic: /mrcd_robot/rs_front/rs_front/color/metadata | Type: realsense2_camera_msgs/msg/Metadata | Count: 29866 | Serialization Format: cdr
                   Topic: /mrcd_robot/cmd_vel_teleop | Type: geometry_msgs/msg/Twist | Count: 19925 | Serialization Format: cdr
                   Topic: /mrcd_robot/zed_node/right_raw/camera_info | Type: sensor_msgs/msg/CameraInfo | Count: 29929 | Serialization Format: cdr
                   Topic: /mrcd_robot/zed_node/pose | Type: geometry_msgs/msg/PoseStamped | Count: 30136 | Serialization Format: cdr
                   Topic: /mrcd_robot/platform/dynamic_joint_states | Type: control_msgs/msg/DynamicJointState | Count: 49810 | Serialization Format: cdr
                   Topic: /mrcd_robot/zed_node/pose_with_covariance | Type: geometry_msgs/msg/PoseWithCovarianceStamped | Count: 30136 | Serialization Format: cdr
                   Topic: /mrcd_robot/zed_node/right_raw/image_raw_color | Type: sensor_msgs/msg/Image | Count: 29929 | Serialization Format: cdr
                   Topic: /mrcd_robot/rs_front/rs_front/color/camera_info | Type: sensor_msgs/msg/CameraInfo | Count: 29866 | Serialization Format: cdr
                   Topic: /mrcd_robot/zed_node/path_map | Type: nav_msgs/msg/Path | Count: 2014 | Serialization Format: cdr
                   Topic: /mrcd_robot/rs_front/rs_front/depth/color/points | Type: sensor_msgs/msg/PointCloud2 | Count: 14942 | Serialization Format: cdr
                   Topic: /mrcd_robot/rs_front/rs_front/depth/camera_info | Type: sensor_msgs/msg/CameraInfo | Count: 14942 | Serialization Format: cdr
                   Topic: /mrcd_robot/platform/odom/filtered | Type: nav_msgs/msg/Odometry | Count: 29888 | Serialization Format: cdr
                   Topic: /mrcd_robot/rs_front/rs_front/color/image_raw | Type: sensor_msgs/msg/Image | Count: 29866 | Serialization Format: cdr
                   Topic: /mrcd_robot/zed_node/imu/mag | Type: sensor_msgs/msg/MagneticField | Count: 50674 | Serialization Format: cdr
[... cut ...]
                   Topic: /mrcd_robot/platform/odom | Type: nav_msgs/msg/Odometry | Count: 49810 | Serialization Format: cdr
                   Topic: /mrcd_robot/rtk_gnss/tcpfix | Type: sensor_msgs/msg/NavSatFix | Count: 4981 | Serialization Format: cdr
                   Topic: /mrcd_robot/rs_front/rs_front/aligned_depth_to_color/camera_info | Type: sensor_msgs/msg/CameraInfo | Count: 14943 | Serialization Format: cdr
```

and the version of the package currently installed:

```
ros-humble-isaac-ros-visual-slam/jammy,now 3.2.5-0jammy amd64 [installed]
```

Maybe you can make some sense of this?