NVIDIA-ISAAC-ROS/isaac_ros_visual_slam #143: Problem with tilted camera mount

sandeepdutta, 2024-01-11:
Hello,
The camera on my robot is mounted pointed upwards . VSLAM seems to be having trouble
with this setup. It looses "Visual Tracking" frequently and creates a map (as visualized in RVIZ)
which looks like it is at an angle to the ground plane. The visual track has a vertical component as well.
I have set "force_planar" to true. Is there something I can tweak to make this work.
It works fine in the same environment when I straighten the camera to point in the direction of motion.

Thank you
Sandeep

swapnesh-wani-nvidia, 2024-02-05:
Could you share the tf tree for the sensors and the base_link? Also, it would help what kind of environment are you working in. If the motion of the robot is not perfectly planar, `force_planar` flag won't help here.

sandeepdutta, 2024-02-06:
Hello Swapnesh,

I have attached the full URDF file for the robot. Of interest are two joints 
a) This is the top camera pointed in the same direction of X   
"\

swapnesh-wani-nvidia, 2024-02-21:
Thanks for the update. Screnshots will be helpful in oerder to diagnose. Could you also share the laucnh files you are using?

sandeepdutta, 2024-03-07:
Hello Swapnesh,

Sorry it took longer than I thought it would. I have run some experiments, and captured some screen shots, note all experiments in the same environment and same starting pose.
a) Camera mounted pointing in direction of travel force_planar_mode = false , all works ok.
[image]

b) Camera mounted with 10 deg upward tilt (this is done to get more visual artifacts) force_planar_mode = false. in the screen shot observe the odometry moving upwards in the same angle as the camera. Once we get some loop closure,
the trajectory is corrected the odometry moves straight for a little while , if no loop closure for a while the odometry will start drifting upwards.
[image]

c) Camera mounted with 10 deg upward tilt force_palar_mode = true. In this case the odometry consistantly moves in the same plane, the plane is tilted by the same amount as the camera tilt.

[image]

With force_planar_mode=true, it seems to be consistent albeit working on a tilted plane . Would be great to get a solution or work around to this problem.
Thank you
Sandeep

sandeepdutta, 2024-03-08:
Hello Swapnesh,

I have some further updates to this issue. I ran two more experiments. I removed the camera tilt from the URDF, i.e.
the camera mount does not include the tilt, the transform thinks that the camera is mounted in the direction of travel.
a) With force_planar_mode=True , the VO generated is correct, it shows the odometry pointing upwards and the
     trajectory is correct (no Z travel). It looks like the VO is able to infer rhat the camera is mounted at an angle .
[image]
b) With force_planar_mode=False, however the odometry travels downwards at the same angle as the camera mount,
so it seems the VO thinks that the camera is moving downwards.
[image]

Best Regards
Sandeep

sandeepdutta, 2024-03-09:
Additionally , "Loadandlocalize" never worked with databases created with tilted camera.
Sandeep