NVIDIA-ISAAC-ROS/isaac_ros_visual_slam #144: [ERROR] X

sandeepdutta, 2024-01-19:
Hello ,

I have started getting these error messages , could you please let me know what I can do to avoid these.
The slam continues to work,  it is able to recover from this error.

Thank you
Sandeep
[component_container-1] [WARN] [1705692918.677522629] [visual_slam_node]: Delta between current and previous frame [133.431397] is above threshold [100.000000]
[component_container-1] [ERROR] X
[component_container-1] [ERROR] X
[component_container-1] [ERROR] X
[component_container-1] [ERROR] X
[component_container-1] [ERROR] X
[component_container-1] [ERROR] X
[component_container-1] [ERROR] X
[component_container-1] [ERROR] X
[component_container-1] [ERROR] X
[component_container-1] [ERROR] X
[component_container-1] [ERROR] X
[component_container-1] [ERROR] X
[component_container-1] [ERROR] X
[component_container-1] [ERROR] X
[component_container-1] [ERROR] X
[component_container-1] [ERROR] X
[component_container-1] [ERROR] X
[component_container-1] [ERROR] X
[component_container-1] [ERROR] X
[component_container-1] [ERROR] X
[component_container-1] [ERROR] X
[component_container-1] [ERROR] X
[component_container-1] [ERROR] X
[component_container-1] [ERROR] X
[component_container-1] [ERROR] X
[component_container-1] [ERROR] X
[component_container-1] [ERROR] X
[component_container-1] [ERROR] X
[component_container-1] [ERROR] X
[component_container-1] [ERROR] X
[component_container-1] [ERROR] X
[component_container-1] [ERROR] X
[component_container-1] [ERROR] X
[component_container-1] [ERROR] X
[component_container-1] [ERROR] X
[component_container-1] [ERROR] X
[component_container-1] [ERROR] X
[component_container-1] [ERROR] X
[component_container-1] [ERROR] X
[component_container-1] [ERROR] X
[component_container-1] [ERROR] X
[component_container-1] [ERROR] X
[component_container-1] [WARN] [1705692939.118423192] [visual_slam_node]: Delta between current and previous frame [133.434571] is above threshold [100.000000]

jaiveersinghNV, 2024-01-29:
Hi @sandeepdutta , 

Could you please identify which launch file you're using? This will help us identify which nodes are running in that component container and isolate where that spurious error message is coming from.

sandeepdutta, 2024-01-30:
Here is the launch file .. I am using a realsense camera .

# SPDX-FileCopyrightText: NVIDIA CORPORATION & AFFILIATES
# Copyright (c) 2021-2023 NVIDIA CORPORATION & AFFILIATES. All rights reserved.
#
# Licensed under the Apache License, Version 2.0 (the "License");
# you may not use this file except in compliance with the License.
# You may obtain a copy of the License at
#
# http://www.apache.org/licenses/LICENSE-2.0
#
# Unless required by applicable law or agreed to in writing, software
# distributed under the License is distributed on an "AS IS" BASIS,
# WITHOUT WARRANTIES OR CONDITIONS OF ANY KIND, either express or implied.
# See the License for the specific language governing permissions and
# limitations under the License.
#
# SPDX-License-Identifier: Apache-2.0

import launch
from launch_ros.actions import ComposableNodeContainer, Node
from launch_ros.descriptions import ComposableNode

def generate_launch_description():

    visual_slam_node = ComposableNode(
        name='visual_slam_node',
        package='isaac_ros_visual_slam',
        plugin='nvidia::isaac_ros::visual_slam::VisualSlamNode',
        parameters=[{
                    'denoise_input_images': True,
                    'rectified_images': True,
                    'enable_debug_mode': False,
                    'debug_dump_path': '/tmp/cuvslam',
                    'enable_slam_visualization': True,
                    'enable_landmarks_view': True,
                    'enable_observations_view': True,
                    'enable_localization_n_mapping':True,
                    'publish_odom_to_base_tf' : True,
                    'publish_map_to_odom_tf': True,
                    'map_frame': 'map',
                    'odom_frame': 'odom',
                    'base_frame': 'base_link',
                    'input_imu_frame': 'IMU_1',
                    'input_base_frame': 'base_link',
                    'image_qos': 'SERVICES_DEFAULT',
                    'input_left_camera_frame': 'rs_top_infra1_frame',
                    'input_right_camera_frame': 'rs_top_infra2_frame',
                    'enable_imu_fusion': False,
                    'gyro_noise_density': 8.6e-05  ,
                    'gyro_random_walk':2.2e-06,
                    'accel_noise_density': 1.4e-03,
                    'accel_random_walk':8.0e-05,
                    'calibration_frequency': 200.0,
                    'img_jitter_threshold_ms': 100.00
                    }],
        remappings=[('stereo_camera/left/image', '/camera/rs_top/infra1/image_rect_raw'),
                    ('stereo_camera/left/camera_info', '/camera/rs_top/infra1/camera_info'),
                    ('stereo_camera/right/image', '/camera/rs_top/infra2/image_rect_raw'),
                    ('stereo_camera/right/camera_info', '/camera/rs_top/infra2/camera_info'),
                    ('visual_slam/imu', '/imu/data')]
    )

    visual_slam_launch_container = ComposableNodeContainer(
        name='visual_slam_launch_container',
        namespace='',
        package='rclcpp_components',
        executable='component_container',
        composable_node_descriptions=[
            visual_slam_node
        ],
        output='screen'
    )

    return launch.LaunchDescription([visual_slam_launch_container])

swapnesh-wani-nvidia, 2024-02-05:
Are there any other launch files that are being launched besides isaac_ros_visual_slam?  Which camera are you using? Error X is not from Isaac ROS packages.

sandeepdutta, 2024-02-06:
Hello Swapnesh,
This particular terminal is running only the visual slam node, with the launch file I provided. The camera is launched from a separate launch file on another terminal. If you see the ERROR & WARNING are coming from the same "[component_container-1]" . I get the error occasionally and it does not seem to effect the performance or quality.

Thank you
Sandeep