NVIDIA-ISAAC-ROS/isaac_ros_visual_slam #145: "Visual tracking is lost" in tutorial

Taeyeon0206, 2024-01-25:
hi,
I was going through the Isaac ROS Vslam tutorial with Isaac Sim.
https://nvidia-isaac-ros.github.io/concepts/visual_slam/cuvslam/tutorial_isaac_sim.html

However, as soon as I launched isaac_ros_visual_slam, "visual tracking is lost" error kept on popping.
I tried echoing the topics published by visual_slam_node, but none of the topics published anything.
I looked at the image_raw topic published by Isaac Sim through rviz2 and it was just fine.
[image]
this is the error message printed.

What could be the problem?

By the way, I am using Ubuntu22.04 and ROS2 humble. I am running both Isaac Sim and Isaac ROS in a separate docker container. I am running Isaac Sim by run headless mode with streaming client. I followed exactly the document "development environment setup" and "isaac ROS Vslam quickstart" and installed isaac_ros_common and isaac_ros_visual_slam in the process.
Also, a strange point is, I tried Isaac ROS map localization and it worked.

Taeyeon0206, 2024-01-30:
Hi @jaiveersinghNV 
I successfully ran the Quickstart. It worked just fine.

1. Fast motion causing the motion blur in the frames
2. Low-lighting conditions.
3. The wrong camerainfo is being published.

These 3 suggestions were in the recommended troubleshooting step. 
The error occurred from the beginning and without any movement, so I thought motion blur is not the case.
Since it is example environment, I thought low-lighting is not the problem either.
Also, I checked the published topic(left and right camera) using rviz2 and it showed up just fine.

Taeyeon0206, 2024-01-31:
maybe this might help.
These are the camera info of left and right camera of front hawk.
left:
[image]
right:
[image]

This is the exact error message that i have been receiving.
[image]

Taeyeon0206, 2024-02-01:
@jaiveersinghNV 
I changed the distortion model into pinhole model after bringing the simulation example environment in Isaac Sim and it worked.
I wonder why the camera distortion model has to be pinhole model. Looking at the visual_slam_impl.cpp code, it seems to me that it has to work in the rational_polynomial distortion model as well.

swapnesh-wani-nvidia, 2024-02-05:
@Taeyeon0206, thanks for bringing this up. Can you verify if the images coming from Isaac Sim's setup are rectified and undistorted or not? If they are rectified and undistorted then it has to be a pinhole camera model.

aravindsairam, 2024-04-26:
@Taeyeon0206 How did you change the distortion model from rational_polynomial to pinhole model ?