NVIDIA-ISAAC-ROS/isaac_ros_visual_slam #148: IMU fusion status

dakejahl, 2024-02-09:
Is there any way to tell if IMU fusion is working or not? The VisualSlamStatus.vo_state doesn't ever change from 1.
```
# Pure visual odometry return code:
# 0 - Unknown state
# 1 - Success
# 2 - Failed
# 3 - Success but invalidated by IMU
uint8 vo_state
```
Same goes for when tracking is lost, `vo_state` remains at 1.

AlexKlimaj, 2024-02-10:
We are testing on a drone, using the output of VIO to hold position with PX4. Knowing the state of would be very helpful in PX4.

https://github.com/ARK-Electronics/isaac_ros-dev

https://www.youtube.com/watch?v=Q3-W-Bj-9Ck

https://review.px4.io/plot_app?log=38789a73-c5d6-4702-bbea-1f87ce57f67a

hemalshahNV, 2024-02-14:
IMU fusion is actually a feature we will ship in the next release of Isaac ROS. The interface had been added for compatibility with other editions of the libcuvslam libraries.

AlexKlimaj, 2024-02-14:
So this statement in the README isn't correct about using IMU?

`This package uses a stereo camera with an IMU to estimate odometry as an input to navigation.`

hemalshahNV, 2024-02-14:
There was an IMU fallback feature (when visual tracking is lost, libcuvslam would try to run on pure IMU for a fix timebound), but that's been replaced with IMU fusion.

hermanndererdmann, 2024-02-29:
is there a timeframe when the new release will be published?

franfrancisco9, 2024-05-23:
Any news on when IMU Fusion will be active?

anthonywebb, 2024-08-08:
It looks like Isaac ROS 3 added it, but I can confirm it still doesnt work

yurii-drevnytskyi-dianomix, 2024-09-20:
Can you confirm that IMU fusion doesn't work? I'm trying to figure out why turning it on doesn't improve the odometry results.