NVIDIA-ISAAC-ROS/isaac_ros_visual_slam #149: Cuda issue when running roslaunch command.

pojolin, 2024-02-12:
I am trying to run the Isaac Ros VSLAM on a Aeeon Boxer 8251AI. It has a Jetson Xavier NX. I followed the Quickstart steps and eveything looks fine until I try running the launch files. I get this error related to Cuda when I do so:

```
ros2 launch isaac_ros_visual_slam isaac_ros_visual_slam.launch.py
[INFO] [launch]: All log files can be found below /home/admin/.ros/log/2024-02-12-18-38-48-978043-v34-gpu-1-437026
[INFO] [launch]: Default logging verbosity is set to INFO
[INFO] [component_container-1]: process started with pid [437037]
[component_container-1] [INFO] [1707763130.098561283] [visual_slam_launch_container]: Load Library: /opt/ros/humble/lib/libvisual_slam_node.so
[component_container-1] [ERROR] [1707763130.109000615] [visual_slam_launch_container]: Failed to load library: Could not load library dlopen error: /usr/local/cuda-11.4/targets/aarch64-linux/lib/libcusolver.so.11: cannot read file data: Is a directory, at /tmp/binarydeb/ros-humble-rcutils-5.1.4/src/shared_library.c:99
[ERROR] [launch_ros.actions.load_composable_nodes]: Failed to load node 'visual_slam_node' of type 'nvidia::isaac_ros::visual_slam::VisualSlamNode' in container '/visual_slam_launch_container': Failed to load library: Could not load library dlopen error: /usr/local/cuda-11.4/targets/aarch64-linux/lib/libcusolver.so.11: cannot read file data: Is a directory, at /tmp/binarydeb/ros-humble-rcutils-5.1.4/src/shared_library.c:99

```

swapnesh-wani-nvidia, 2024-02-21:
Looks like you are running this with an old version of Jetpack. Could you share the Jetpack version?

pojolin, 2024-02-21:
Yes I'm sorry, I realized this later but couldn't change the Jetpack version on the device I was using. Tested on some other device with the required prerequisites and it worked. 

Thank you.

closed 2024-02-21 by pojolin