NVIDIA-ISAAC-ROS/isaac_ros_visual_slam #150: Feature request : localization only mode.

sandeepdutta, 2024-02-15:
Hello,

Would be great if the SLAM could be made to run in "localization" mode only.
That is , it will not publish any vo_pose or odom->base_link if it cannot localize .
If there is way to do this already then we will be happy to try.

Thank you
Sandeep

swapnesh-wani-nvidia, 2024-02-21:
Could you clarify this? Are you looking to localize on a different map than the one created by Isaac ROS Visual SLAM? 

> That is , it will not publish any vo_pose or odom->base_link if it cannot localize .

sandeepdutta, 2024-02-21:
Hello Swapnesh,

We are trying to localize in the map generated by VSLAM. We are  using EKF to fuse wheel odometry & visual odometry, in a
premapped environment. If  "Visual Tracking Lost" , the VSLAM algorithm reverts to "mapping" mode if it finds no loop closure and continues to publish visual_odometry with low "covariance" this confuses the EKF filter to rely on VSLAM more than it should.  We are planning to update the "visual odometry" (set_odometry_pose) from the EKF fused odometry if visual odometry is lost.
If we had a "localize" only mode, then it would make the fusion much easier.

Thank you
Sandeep