NVIDIA-ISAAC-ROS/isaac_ros_visual_slam #151: isaac_ros_visual_slam_zed.launch.py If an error occurs when running

konghun, 2024-02-19:
**Hello, I am trying to do vslam using ZED2i.

I've done all the environment settings, but I'm getting this error. Does anyone have a similar experience?**

[ERROR] [launch_ros.actions.load_composable_nodes]: Failed to load node 'image_format_node_left' of type 'nvidia::isaac_ros::image_proc::ImageFormatConverterNode' in container '/visual_slam_launch_container': Could not find requested resource in ament index
[ERROR] [launch_ros.actions.load_composable_nodes]: Failed to load node 'image_format_node_right' of type 'nvidia::isaac_ros::image_proc::ImageFormatConverterNode' in container '/visual_slam_launch_container': Could not find requested resource in ament index

**visual_slam_node is said to have been successfully loaded within the container**

[INFO] [launch_ros.actions.load_composable_nodes]: Loaded node '/visual_slam_node' in container '/visual_slam_launch_container'

swapnesh-wani-nvidia, 2024-02-21:
Looks like you are missing `ImageFormatConverterNode`. It is part of `isaac_ros_image_proc` [package](https://github.com/NVIDIA-ISAAC-ROS/isaac_ros_image_pipeline/tree/main/isaac_ros_image_proc). Make sure you are building the package correctly. Could you share how you built the package?

Jimi1811, 2024-04-29:
Hi there! I encountered a similar issue while working with the documentation for [Isaac ROS VSLAM](https://nvidia-isaac-ros.github.io/concepts/visual_slam/cuvslam/tutorial_zed.html). The documentation suggests installing the package using `sudo apt-get install -y ros-humble-isaac-ros-stereo-image-proc`. Despite following this instruction, I faced the same problem.

In an attempt to resolve it, I moved the package located in `/opt/ros/humble/share` to the `src` directory. However, upon compilation, I encountered an error indicating the absence of a `.txt` file. Consequently, I decided to clone the package from [this repository](https://github.com/NVIDIA-ISAAC-ROS/isaac_ros_image_pipeline/tree/main) and compiled it successfully.

After compilation, I verified its presence using the command `ros2 pkg list`, only to find that the output remained unchanged, unfortunately.

```
admin@ubuntu:/workspaces/isaac_ros-dev$ ros2 launch isaac_ros_visual_slam isaac_ros_visual_slam_zed.launch.py
[INFO] [launch]: All log files can be found below /home/admin/.ros/log/2024-04-29-17-39-30-412315-ubuntu-20695
[INFO] [launch]: Default logging verbosity is set to INFO
[INFO] [component_container-1]: process started with pid [20706]
[component_container-1] [ERROR] [1714430376.086975535] [visual_slam_launch_container]: Could not find requested resource in ament index
[ERROR] [launch_ros.actions.load_composable_nodes]: Failed to load node 'image_format_node_left' of type 'nvidia::isaac_ros::image_proc::ImageFormatConverterNode' in container '/visual_slam_launch_container': Could not find requested resource in ament index
[component_container-1] [ERROR] [1714430376.093385568] [visual_slam_launch_container]: Could not find requested resource in ament index
[ERROR] [launch_ros.actions.load_composable_nodes]: Failed to load node 'image_format_node_right' of type 'nvidia::isaac_ros::image_proc::ImageFormatConverterNode' in container '/visual_slam_launch_container': Could not find requested resource in ament index
[component_container-1] [INFO] [1714430376.097873189] [visual_slam_launch_container]: Load Library: /workspaces/isaac_ros-dev/install/isaac_ros_visual_slam/lib/libvisual_slam_node.so
[component_container-1] [INFO] [1714430376.328403243] [visual_slam_launch_container]: Found class: rclcpp_components::NodeFactoryTemplate<nvidia::isaac_ros::visual_slam::VisualSlamNode>
[component_container-1] [INFO] [1714430376.328525005] [visual_slam_launch_container]: Instantiate class: rclcpp_components::NodeFactoryTemplate<nvidia::isaac_ros::visual_slam::VisualSlamNode>
[component_container-1] [INFO] [1714430376.419255316] [visual_slam_node]: cuVSLAM version: 11.4
[component_container-1] [INFO] [1714430379.014976567] [visual_slam_node]: Time taken by CUVSLAM_WarmUpGPU(): 2.595536
[INFO] [launch_ros.actions.load_composable_nodes]: Loaded node '/visual_slam_node' in container '/visual_slam_launch_container'
```

I recently encountered this issue, which stemmed from a problem with policy in the subscribers. To address it, I took the following steps: First, I commented out the zed node in `ros2 launch isaac_ros_visual_slam isaac_ros_visual_slam_zed.launch.py`. Next, I ran `ros2 launch zed_wrapper zed_camera.launch.py camera_model:=zed2i`.

Muhammad540, 2024-12-30:
> **Hello, I am trying to do vslam using ZED2i.
>
> I've done all the environment settings, but I'm getting this error. Does anyone have a similar experience?**
>
> [ERROR] [...]

were you able to resolve this ?

Muhammad540, 2024-12-30:
> Hi there! I encountered a similar issue while working with the documentation for [Isaac ROS VSLAM](https://nvidia-isaac-ros.github.io/concepts/visual_slam/cuvslam/tutorial_zed.html). The [...]

did this solve your issue ?