NVIDIA-ISAAC-ROS/isaac_ros_visual_slam #152: No messages being published

brian7989, 2024-02-26:
I am facing an issue where I am unable to get any messages from `ros2 topic echo /visual_slam/tracking/odometry`. (Or any other `/visual_slam/tracking/*` topics). This is strange because I am able to see `publisher count = 1` but no messages are actually being published. I would appreciate any help or insights on how to resolve this issue so that I can get messages to be properly published.

### Background

I am using Jetson Orin Nano 8gb Developer Kit. I have followed [this guide](https://nvidia-ai-iot.github.io/jetson_isaac_ros_visual_slam_tutorial/quickstart.html) step by step and have flashed the provided custom SD card for Jetson Orin Nano. I am using Intel Realsense D455, which is a compatible model, and I can see the camera images when I run `realsense-viewer`.

### Output of `ros2 launch isaac_ros_visual_slam isaac_ros_visual_slam_realsense.launch.py`:

```
[INFO] [launch]: All log files can be found below /home/admin/.ros/log/2024-02-26-10-47-07-906447-jon-jp511-isaac-ros-13087
[INFO] [launch]: Default logging verbosity is set to INFO
[INFO] [component_container-1]: process started with pid [13099]
[INFO] [realsense2_camera_node-2]: process started with pid [13101]
[component_container-1] [INFO] [1708973228.580282805] [visual_slam_launch_container]: Load Library: /workspaces/isaac_ros-dev/install/isaac_ros_visual_slam/lib/libvisual_slam_node.so
[realsense2_camera_node-2] [INFO] [1708973228.797161635] [camera.camera]: RealSense ROS v4.51.1
[realsense2_camera_node-2] [INFO] [1708973228.797521075] [camera.camera]: Built with LibRealSense v2.53.1
[realsense2_camera_node-2] [INFO] [1708973228.797554005] [camera.camera]: Running with LibRealSense v2.53.1
[component_container-1] [INFO] [1708973228.907512002] [visual_slam_launch_container]: Found class: rclcpp_components::NodeFactoryTemplate<isaac_ros::visual_slam::VisualSlamNode>
[component_container-1] [INFO] [1708973228.908315655] [visual_slam_launch_container]: Instantiate class: rclcpp_components::NodeFactoryTemplate<isaac_ros::visual_slam::VisualSlamNode>
[realsense2_camera_node-2] [INFO] [1708973228.973460239] [camera.camera]: Device with serial number 038122250330 was found.
[realsense2_camera_node-2] 
[realsense2_camera_node-2] [INFO] [1708973228.974007176] [camera.camera]: Device with physical ID /sys/devices/platform/3610000.xhci/usb1/1-2/1-2.1/1-2.1:1.0/video4linux/video0 was found.
[realsense2_camera_node-2] [INFO] [1708973228.974044137] [camera.camera]: Device with name Intel RealSense D455 was found.
[realsense2_camera_node-2] [INFO] [1708973228.975533710] [camera.camera]: Device with port number 1-2.1 was found.
[realsense2_camera_node-2] [INFO] [1708973228.975975490] [camera.camera]: Device USB type: 2.1
[realsense2_camera_node-2] [WARN] [1708973228.976019748] [camera.camera]: Device 038122250330 is connected using a 2.1 port. Reduced performance is expected.
[realsense2_camera_node-2] [INFO] [1708973229.021068531] [camera.camera]: getParameters...
[realsense2_camera_node-2] [INFO] [1708973229.026116378] [camera.camera]: JSON file is not provided
[realsense2_camera_node-2] [INFO] [1708973229.026245408] [camera.camera]: Device Name: Intel RealSense D455
[realsense2_camera_node-2] [INFO] [1708973229.026277217] [camera.camera]: Device Serial No: 038122250330
[realsense2_camera_node-2] [INFO] [1708973229.026303298] [camera.camera]: Device physical port: /sys/devices/platform/3610000.xhci/usb1/1-2/1-2.1/1-2.1:1.0/video4linux/video0
[realsense2_camera_node-2] [INFO] [1708973229.026322371] [camera.camera]: Device FW version: 05.12.06.00
[realsense2_camera_node-2] [INFO] [1708973229.026339396] [camera.camera]: Device Product ID: 0x0B5C
[realsense2_camera_node-2] [INFO] [1708973229.026356933] [camera.camera]: Sync Mode: Off
[component_container-1] [INFO] [1708973229.108896135] [visual_slam_node]: Elbrus version: 10.5
[INFO] [launch_ros.actions.load_composable_nodes]: Loaded node '/visual_slam_node' in container '/visual_slam_launch_container'
[realsense2_camera_node-2] [ERROR] [1708973229.156535340] [camera.camera]: Given value, 640x360x90 is invalid. Set ROS param back to: 640x480x15
[realsense2_camera_node-2] [WARN] [1708973229.178067429] [camera.camera]: Could not set param: rgb_camera.power_line_frequency with 3 Range: [0, 2]: parameter 'rgb_camera.power_line_frequency' could not be set: Parameter {rgb_camera.power_line_frequency} doesn't comply with integer range.
[realsense2_camera_node-2] [INFO] [1708973229.307642672] [camera.camera]: Stopping Sensor: Stereo Module
[realsense2_camera_node-2] [INFO] [1708973229.457460666] [camera.camera]: Starting Sensor: Stereo Module
[realsense2_camera_node-2] [INFO] [1708973229.545278958] [camera.camera]: Open profile: stream_type: Infra(1), Format: Y8, Width: 640, Height: 480, FPS: 15
[realsense2_camera_node-2] [INFO] [1708973229.554728575] [camera.camera]: Stopping Sensor: Motion Module
[realsense2_camera_node-2] [INFO] [1708973229.640988171] [camera.camera]: Starting Sensor: Motion Module
[realsense2_camera_node-2] [INFO] [1708973229.718410915] [camera.camera]: Open profile: stream_type: Accel(0)Format: MOTION_XYZ32F, FPS: 63
[realsense2_camera_node-2] [INFO] [1708973229.718666287] [camera.camera]: Open profile: stream_type: Gyro(0)Format: MOTION_XYZ32F, FPS: 200
[realsense2_camera_node-2] [INFO] [1708973229.743289782] [camera.camera]: RealSense Node Is Up!
[realsense2_camera_node-2] [WARN] [1708973229.783953371] [camera.camera]:
```
(The only red flag I see is `Could not set param: rgb_camera.power_line_frequency with 3 Range: [0, 2]: parameter 'rgb_camera.power_line_frequency' could not be set: Parameter {rgb_camera.power_line_frequency} doesn't comply with integer range.`, but I am not sure if this is the core issue)

### Output of `ros2 wtf --report`:
```
topic [type]            : /visual_slam/tracking/odometry [nav_msgs/msg/Odometry]
publisher node          : visual_slam_node
subscriber node         : _ros2cli_13486
compatibility status    : OK
...
topic               : /visual_slam/tracking/odometry
publisher count     : 1
subscriber count    : 0
topic               : /visual_slam/tracking/slam_path
publisher count     : 1
subscriber count    : 0
...
```

### Additional information:
- I have tried stopping and starting the daemon
- I tried recreating the Docker container (exiting and running `isaac_ros_container` again)
- Power-mode is set to 15W for the Jetson
- I can read from `/camera/*` topics, but the average rate is a little slow: `ros2 topic hz /camera/infra1/image_rect_raw --window 20
average rate: 12.309 min: 0.055s max: 0.200s std dev: 0.03933s window: 13`

brian7989, 2024-02-26:
- I've also tried removing `/build` `/log` `/install` and tried `colcon build --symlink-install` but the issue persists.
- Another potential red flag is the outdated Realsense firmware. There is a message in realsense-viewer that says "Intel RealSense D455, Current Version: 05.12.06.00, Recommended Version: 5.14.0.0". However, whenever I try to install an update, it crashes mid-installation.
- I ran `ros2 run tf2_tools view_frames` but the frames look a little odd: 
[frames_2024-02-26_12.34.47.pdf](https://github.com/NVIDIA-ISAAC-ROS/isaac_ros_visual_slam/files/14411024/frames_2024-02-26_12.34.47.pdf)

jaiveersinghNV, 2024-02-28:
Hi @brian7989 ,

I believe we're already working with you on [this thread](https://forums.developer.nvidia.com/t/isaac-ros-vslam-no-messages-being-published/284103/2) in the Isaac ROS Forum.

To reiterate the main point of that reply here, it looks like you're using an outdated tutorial. Please reference the official [Isaac ROS Docs Site](https://NVIDIA-ISAAC-ROS.github.io) to get the latest information and quickstart guides.

brian7989, 2024-02-28:
Yes I'll continue on the ROS forum. Thanks.

closed 2024-02-28 by brian7989