NVIDIA-ISAAC-ROS/isaac_ros_visual_slam #155: LoadMapAdnLocalize crashes

sandeepdutta, 2024-03-27:
Hello,

The service call to LoadMapAndLocalize sometimes crashes. Below is the error messages we
get , restarting and reissuing the service call will work. Doing a quick google search, it looks
like it could be caused by a  "race" condition.

Please let me know if we can provide any more information
Best Regards 
Sandeep
[component_container_mt-2] [WARN] [1711573751.678238729] [visual_slam_node]: CallbackLoadMapAndLocalizeAccepted()
[component_container_mt-2] [INFO] [1711573751.678336682] [visual_slam_node]: cuvslam: LoadMapAndLocalize /workspaces/isaac_ros-dev/src/jeeves_navigation/maps/visual_map [0.000000, 0.000000, 0.000000]
[component_container_mt-2] terminate called after throwing an instance of 'std::runtime_error'
[component_container_mt-2]   what():  Taking data from action server but nothing is ready
[ERROR] [component_container_mt-2]: process has died [pid 92842, exit code -6, cmd '/opt/ros/humble/lib/rclcpp_components/component_container_mt --ros-args -r __node:=vslam_container'].

harborhoffer, 2024-04-11:
Also periodically getting the same error when loading a map

```
[component_container_mt-1] [WARN] [1712759840.096308135] [visual_slam_node]: CallbackLoadMapAndLocalizeAccepted()
[component_container_mt-1] [INFO] [1712759840.096384712] [visual_slam_node]: cuvslam: LoadMapAndLocalize /workspaces/isaac_ros-dev/vslam-maps/office-4-9 [0.000000, 0.000000, 0.000000]
[component_container_mt-1] terminate called after throwing an instance of 'std::runtime_error'
[component_container_mt-1]   what():  Taking data from action server but nothing is ready
[ERROR] [component_container_mt-1]: process has died [pid 14815, exit code -6, cmd '/opt/ros/humble/lib/rclcpp_components/component_container_mt --ros-args -r __node:=shared_nvblox_container'].
```

sandeepdutta, 2024-04-11:
Seems to happen with high system load. I have nav2, nvblox , vslam & controller software running on an
ORIN Agx (32GB) and it happens very frequently.
Any workaround suggestion will be greatly appreciated.
Sandeep

JCastel358, 2024-05-09:
Hello, Im having same issue that it crashes but different error:
```
between current and previous frame [22.254848] is above threshold [20.000000]
[component_container-1] [WARN] [1715217219.969362526] [visual_slam_node]: CallbackLoadMapAndLocalizeAccepted()
[component_container-1] [INFO] [1715217219.969472736] [visual_slam_node]: cuvslam: LoadMapAndLocalize /workspaces/isaac_ros-dev/maps/lab_map_vslam/data.mbd [0.000000, 0.000000, 0.000000]
[component_container-1] [WARN] [1715217219.975820676] [visual_slam_node]: Localization failed. Status=4
[component_container-1] [WARN] [1715217219.977607977] [visual_slam_node]: CUVSLAM_LocalizeInExistDb Error 4
[component_container-1] terminate called after throwing an instance of 'rclcpp::exceptions::RCLError'
[component_container-1]   what():  goal_handle attempted invalid transition from state ABORTED with event ABORT, at /tmp/binarydeb/ros-humble-rcl-action-5.3.6/src/rcl_action/goal_handle.c:95
[ERROR] [component_container-1]: process has died [pid 24714, exit code -6, cmd '/opt/ros/humble/lib/rclcpp_components/component_container --ros-args -r __node:=visual_slam_launch_container -r __ns:=/'].

```

retazo0018, 2024-08-19:
Hi @hemalshahNV,

I also observe the same issue in Orin. Please provide support.

Thanks,

hemalshahNV, 2024-08-19:
We're taking a look to see if we can reproduce this. We may have to find it via code inspection if it is a subtle race condition. Thanks for reporting this issue!

ahmadalahmar, 2024-09-15:
Hi @hemalshahNV 
Any updates about this bug. I am currently working on jetson Orin nano with D435i camera and I am having the same issue with LoadMapAndLocalize. the following error is the same as @JCastel358 :

[component_container-1] [WARN] [1726395004.147950147] [visual_slam_node]: CallbackLoadMapAndLocalizeAccepted()
[component_container-1] [INFO] [1726395004.148065254] [visual_slam_node]: cuvslam: LoadMapAndLocalize /home/invdro/maps/data.mbd [0.000000, 0.000000, 0.000000]
[component_container-1] [WARN] [1726395004.148366157] [visual_slam_node]: Localization failed. Status=6
[component_container-1] [WARN] [1726395004.148400333] [visual_slam_node]: CUVSLAM_LocalizeInExistDb Error 6
[component_container-1] terminate called after throwing an instance of 'rclcpp::exceptions::RCLError'
[component_container-1]   what():  goal_handle attempted invalid transition from state ABORTED with event ABORT, at ./src/rcl_action/goal_handle.c:95
[ERROR] [component_container-1]: process has died [pid 50396, exit code -6, cmd '/opt/ros/humble/lib/rclcpp_components/component_container --ros-args -r __node:=visual_slam_launch_container -r __ns:=/'].

bahramii, 2024-10-08:
in invidia forum:

https://forums.developer.nvidia.com/t/isaac-ros-visual-slam-loadmapadnlocalize-crashes/292250/4

the moderator says that loading map should be before vslam node starts mapping.

how is this possibe to do that?