NVIDIA-ISAAC-ROS/isaac_ros_visual_slam #160: problem with colcon build

raffy952, 2024-05-14:
I have a problem to build and source the workspace inside the container, when i execute the following commands :
`cd /workspaces/isaac_ros-dev && \
  colcon build --symlink-install && \
  source install/setup.bash`

I receive the following output:
  ```
Traceback (most recent call last):
  File "/usr/bin/colcon", line 33, in <module>
    sys.exit(load_entry_point('colcon-core==0.15.1', 'console_scripts', 'colcon')())
  File "/usr/lib/python3/dist-packages/colcon_core/command.py", line 118, in main
    return _main(command_name=command_name, argv=argv)
  File "/usr/lib/python3/dist-packages/colcon_core/command.py", line 185, in _main
    create_log_path(args.verb_name)
  File "/usr/lib/python3/dist-packages/colcon_core/location.py", line 186, in create_log_path
    os.makedirs(str(path))
  File "/usr/lib/python3.8/os.py", line 223, in makedirs
    mkdir(name, mode)
PermissionError: [Errno 13] Permission denied: 'log/build_2024-05-14_14-21-54'
```

If i execute the commands with sudo i receive the following output:
```
Starting >>> isaac_ros_common
Starting >>> realsense2_camera_msgs
Starting >>> isaac_ros_test
--- stderr: isaac_ros_common                                          
CMake Error at CMakeLists.txt:25 (find_package):
  By not providing "Findament_cmake_auto.cmake" in CMAKE_MODULE_PATH this
  project has asked CMake to find a package configuration file provided by
  "ament_cmake_auto", but CMake did not find one.

  Could not find a package configuration file provided by "ament_cmake_auto"
  with any of the following names:

 ament_cmake_autoConfig.cmake
 ament_cmake_auto-config.cmake

  Add the installation prefix of "ament_cmake_auto" to CMAKE_PREFIX_PATH or
  set "ament_cmake_auto_DIR" to a directory containing one of the above
  files.  If "ament_cmake_auto" provides a separate development package or
  SDK, be sure it has been installed.

Failed   <<< isaac_ros_common [0.26s, exited with code 1]
Aborted  <<< realsense2_camera_msgs [0.25s]
Aborted  <<< isaac_ros_test [0.50s]                      

Summary: 0 packages finished [1.31s]
  1 package failed: isaac_ros_common
  2 packages aborted: isaac_ros_test realsense2_camera_msgs
  2 packages had stderr output: isaac_ros_common realsense2_camera_msgs
  10 packages not processed
```

vyeevani, 2025-02-14:
did you ever figure out what went wrong? having the same issue

TomaszTB, 2025-09-02:
I ran into this issue and solved it by running `source /opt/ros/humble/setup.bash` before `colcon build`