NVIDIA-ISAAC-ROS/isaac_ros_visual_slam #164: Support for Zed cameras under Isaac ROS version 3 with JP 6

jwknaup, 2024-06-05:
Are there plans to officially support the Zed cameras again now that Stereolabs has released an update to their SDK that supports JetPack 6? 

In the meantime, would I be better off trying to setup the integration with the Zed cameras myself or roll back to the previous version of Isaac ROS (2.1) and the Zed SDK using JetPack 5.1?

https://nvidia-isaac-ros.github.io/releases/index.html#limitations

Thanks!

gordongrigor, 2024-06-06:
It's better to remain on Jetpack 5.x until StereoLabs updates the software for Jetpack 6. StereoLabs is working on this, and has commented on updating to Jetpack 6 in their forums.

jwknaup, 2024-06-06:
Hi, thanks for the reply!

It looks like stereolabs has released support for Jetpack 6 (just a couple weeks ago). https://www.stereolabs.com/developers/release#412-7f794d6a747d

jwknaup, 2024-06-14:
I was able to get it working using the latest versions of JetPack, Isaac ROS Visual SLAM, Zed SDK, and Zed ROS2 Wrapper.

In case anyone is curious, here are the steps I followed, with the launch file attached. I originally tried using the older versions of everything with the Isaac docker container, but ran into issues and found it was easier to work from the ground up with the latest versions. Cheers!

1. Install Jetpack 6.0 (latest) (I am using an Orin Nano dev kit with an nvme SSD as the sole storage device)
2. Complete NVIDIA [Compute Setup](https://nvidia-isaac-ros.github.io/getting_started/hardware_setup/compute/index.html)
3. Setup NVIDIA [Isaac ROS apt repositories and install ROS2 Humble](https://nvidia-isaac-ros.github.io/getting_started/isaac_apt_repository.html)
4. Install Zed [SDK 4.1 (latest) for Jetpack 6.0](https://www.stereolabs.com/developers/release)
5. Clone and build the (latest - master) [Zed ROS2 packages](https://www.stereolabs.com/docs/ros2#build-the-package)
6. Install the prebuilt Isaac ROS Visual SLAM package 
`sudo apt-get install ros-humble-isaac-ros-visual-slam`
7. Create a [custom ROS2 package](https://docs.ros.org/en/humble/Tutorials/Beginner-Client-Libraries/Creating-Your-First-ROS2-Package.html) with the [attached launch file](https://github.com/user-attachments/files/15843567/visual_slam.launch.py.txt) (be sure to remove the .txt extension). I filled in the parameters (e.g. topics and transform frames) using the info in the [visual_slam API](https://nvidia-isaac-ros.github.io/repositories_and_packages/isaac_ros_visual_slam/isaac_ros_visual_slam/index.html#api) and [Zed Node API](https://www.stereolabs.com/docs/ros2/zed-node)
8. Launch Zed Node
`ros2 launch zed_wrapper zed_camera.launch.py camera_model:={ZED_MODEL}`
9. Launch Isaac ROS Visual SLAM
`ros2 launch {CUSTOM_PACKAGE} {ATTACHED_LAUNCH_FILE}`

immanueln98, 2024-06-23:
> I was able to get it working using the latest versions of JetPack, Isaac ROS Visual SLAM, Zed SDK, and Zed ROS2 Wrapper.
>
> In case anyone is curious, here are the steps I followed, with the launch [...]

I seem not to be able to locate the package from the apt repo as shown below

`sudo apt-get install -y ros-humble-isaac-ros-visual-slam
Reading package lists... Done
Building dependency tree... Done
Reading state information... Done
E: Unable to locate package ros-humble-isaac-ros-visual-slam
`

jwknaup, 2024-06-24:
Are you sure all the steps to add the NVIDIA Isaac repositories were successful and that you are using a supported architecture and OS? Seems like something must have gone wrong there.

hermanndererdmann, 2024-06-30:
@jwknaup did you install the SDK in the docker container or outside? I have it outside and cannot install it inside.
 when i try to build the ROS package inside the container, i get this:

Finished <<< zed_interfaces [47.2s]
Starting >>> zed_components
--- stderr: zed_components                            
CMake Error at CMakeLists.txt:81 (find_package):
  By not providing "FindZED.cmake" in CMAKE_MODULE_PATH this project has
  asked CMake to find a package configuration file provided by "ZED", but
  CMake did not find one.

  Could not find a package configuration file provided by "ZED" with any of
  the following names:

    ZEDConfig.cmake
    zed-config.cmake

  Add the installation prefix of "ZED" to CMAKE_PREFIX_PATH or set "ZED_DIR"
  to a directory containing one of the above files.  If "ZED" provides a
  separate development package or SDK, be sure it has been installed.

---
Failed   <<< zed_components [0.42s, exited with code 1]

looks like it doesn't see the installed SDK
what did i do wrong?

jwknaup, 2024-07-01:
I did not use docker. I installed everything natively

hermanndererdmann, 2024-07-02:
okay i tried that now i have this error.
```
Finished <<< isaac_ros_visual_slam_interfaces [11.6s]                        
Starting >>> isaac_ros_visual_slam
--- stderr: isaac_ros_visual_slam                                 
In file included from /usr/include/cuda/std/cmath:19,
                 from /usr/include/cuda/std/complex:12,
                 from /opt/ros/humble/share/isaac_ros_gxf/gxf/include/gxf/std/complex.hpp:23,
                 from /opt/ros/humble/share/isaac_ros_gxf/gxf/include/gxf/std/tensor.hpp:31,
                 from /opt/ros/humble/share/isaac_ros_gxf/gxf/include/gxf/multimedia/video.hpp:30,
                 from /opt/ros/humble/include/isaac_ros_nitros_image_type/nitros_image_view.hpp:27,
                 from /home/ubuntu/nvidia_slam/src/isaac_ros_visual_slam/isaac_ros_visual_slam/include/isaac_ros_visual_slam/impl/types.hpp:27,
                 from /home/ubuntu/nvidia_slam/src/isaac_ros_visual_slam/isaac_ros_visual_slam/include/isaac_ros_visual_slam/impl/viz_helper.hpp:27,
                 from /home/ubuntu/nvidia_slam/src/isaac_ros_visual_slam/isaac_ros_visual_slam/include/isaac_ros_visual_slam/impl/landmarks_vis_helper.hpp:24,
                 from /home/ubuntu/nvidia_slam/src/isaac_ros_visual_slam/isaac_ros_visual_slam/src/impl/landmarks_vis_helper.cpp:24:
/usr/include/cuda/std/detail/libcxx/include/cmath:569:101: error: ‘float cuda::std::__4::hypot(float, float, float)’ conflicts with a previous declaration
  569 | inline _LIBCUDACXX_INLINE_VISIBILITY float       hypot(       float x,       float y,       float z ) { return sqrt(x*x + y*y + z*z); }
      |                                                                                                     ^
In file included from /usr/include/c++/11/random:38,
                 from /home/ubuntu/nvidia_slam/src/isaac_ros_visual_slam/isaac_ros_visual_slam/src/impl/landmarks_vis_helper.cpp:20:
/usr/include/c++/11/cmath:1868:3: note: previous declaration ‘float std::hypot(float, float, float)’
 1868 |   hypot(float __x, float __y, float __z)
      |   ^~~~~
In file included from /usr/include/cuda/std/cmath:19,
                 from /usr/include/cuda/std/complex:12,
                 from /opt/ros/humble/share/isaac_ros_gxf/gxf/include/gxf/std/complex.hpp:23,
                 from /opt/ros/humble/share/isaac_ros_gxf/gxf/include/gxf/std/tensor.hpp:31,
                 from /opt/ros/humble/share/isaac_ros_gxf/gxf/include/gxf/multimedia/video.hpp:30,
                 from /opt/ros/humble/include/isaac_ros_nitros_image_type/nitros_image_view.hpp:27,
                 from /home/ubuntu/nvidia_slam/src/isaac_ros_visual_slam/isaac_ros_visual_slam/include/isaac_ros_visual_slam/impl/types.hpp:27,
                 from /home/ubuntu/nvidia_slam/src/isaac_ros_visual_slam/isaac_ros_visual_slam/include/isaac_ros_visual_slam/impl/viz_helper.hpp:27,
                 from /home/ubuntu/nvidia_slam/src/isaac_ros_visual_slam/isaac_ros_visual_slam/include/isaac_ros_visual_slam/impl/landmarks_vis_helper.hpp:24,
                 from /home/ubuntu/nvidia_slam/src/isaac_ros_visual_slam/isaac_ros_visual_slam/src/impl/landmarks_vis_helper.cpp:24:
/usr/include/cuda/std/detail/libcxx/include/cmath:570:101: error: ‘double cuda::std::__4::hypot(double, double, double)’ conflicts with a previous declaration
  570 | inline _LIBCUDACXX_INLINE_VISIBILITY double      hypot(      double x,      double y,      double z ) { return sqrt(x*x + y*y + z*z); }
      |                                                                                                     ^
In file included from /usr/include/c++/11/random:38,
                 from /home/ubuntu/nvidia_slam/src/isaac_ros_visual_slam/isaac_ros_visual_slam/src/impl/landmarks_vis_helper.cpp:20:
/usr/include/c++/11/cmath:1872:3: note: previous declaration ‘double std::hypot(double, double, double)’
 1872 |   hypot(double __x, double __y, double __z)
      |   ^~~~~
In file included from /usr/include/cuda/std/cmath:19,
                 from /usr/include/cuda/std/complex:12,
                 from /opt/ros/humble/share/isaac_ros_gxf/gxf/include/gxf/std/complex.hpp:23,
                 from /opt/ros/humble/share/isaac_ros_gxf/gxf/include/gxf/std/tensor.hpp:31,
                 from /opt/ros/humble/share/isaac_ros_gxf/gxf/include/gxf/multimedia/video.hpp:30,
                 from /opt/ros/humble/include/isaac_ros_nitros_image_type/nitros_image_view.hpp:27,
                 from /home/ubuntu/nvidia_slam/src/isaac_ros_visual_slam/isaac_ros_visual_slam/include/isaac_ros_visual_slam/impl/types.hpp:27,
                 from /home/ubuntu/nvidia_slam/src/isaac_ros_visual_slam/isaac_ros_visual_slam/include/isaac_ros_visual_slam/impl/viz_helper.hpp:27,
                 from /home/ubuntu/nvidia_slam/src/isaac_ros_visual_slam/isaac_ros_visual_slam/include/isaac_ros_visual_slam/impl/landmarks_vis_helper.hpp:24,
                 from /home/ubuntu/nvidia_slam/src/isaac_ros_visual_slam/isaac_ros_visual_slam/src/impl/landmarks_vis_helper.cpp:24:
/usr/include/cuda/std/detail/libcxx/include/cmath:572:101: error: ‘long double cuda::std::__4::hypot(long double, long double, long double)’ conflicts with a previous declaration
  572 | inline _LIBCUDACXX_INLINE_VISIBILITY long double hypot( long double x, long double y, long double z ) { return sqrt(x*x + y*y + z*z); }
      |                                                                                                     ^
In file included from /usr/include/c++/11/random:38,
[... cut ...]
gmake[2]: *** [CMakeFiles/visual_slam_node.dir/build.make:146: CMakeFiles/visual_slam_node.dir/src/impl/posegraph_vis_helper.cpp.o] Error 1
gmake[2]: *** [CMakeFiles/visual_slam_node.dir/build.make:160: CMakeFiles/visual_slam_node.dir/src/impl/visual_slam_impl.cpp.o] Error 1
gmake[2]: *** [CMakeFiles/visual_slam_node.dir/build.make:76: CMakeFiles/visual_slam_node.dir/src/visual_slam_node.cpp.o] Error 1
gmake[1]: *** [CMakeFiles/Makefile2:163: CMakeFiles/visual_slam_node.dir/all] Error 2
gmake: *** [Makefile:146: all] Error 2
---
Failed   <<< isaac_ros_visual_slam [25.8s, exited with code 2]
```

jwknaup, 2024-07-02:
Are you trying to build visual_slam from source? I just installed the prebuilt package. See step 6.

hermanndererdmann, 2024-07-22:
got it to work, thank you.
do you also have issues with the imu? sometimes it just jumpes somewhere and continues from there