NVIDIA-ISAAC-ROS/isaac_ros_visual_slam #180: problem of isaac_ros_visual_slam_dual_realsense_fusion.launch.py

silence-moon, 2024-10-22:
Thanks for your great work. 
hardware: Jetson-AGX Xavier 
platform: docker
isaac_ros_visual_slam version: dp3.1
description:
I can launch single  realsense camera on slam system. However, when I try to launch the file  isaac_ros_visual_slam_dual_realsense_fusion.launch.py , I find there is no slam outputs compared to single camera. The function VisualSlamNode::VisualSlamImpl::Init on visual_slam_impl.cpp cannot run. And callback function TrackCameraPose  also cannot run. 
It stoped output after this messages:

[realsense2_camera_node-5] [INFO] [1729583953.606905347] [camera_back.camera_back]: RealSense Node Is Up!
[component_container-4] [INFO] [1729583959.401821813] [visual_slam_front.visual_slam_front_node]: Time taken by CUVSLAM_WarmUpGPU(): 6.402246
[component_container-4] [INFO] [1729583959.441170315] [visual_slam_front.visual_slam_front_node]: VisualSlamNode end.
[INFO] [launch_ros.actions.load_composable_nodes]: Loaded node '/visual_slam_front/visual_slam_front_node' in container '/visual_slam_front_launch_container'
[component_container-3] [INFO] [1729583959.957228420] [visual_slam_back.visual_slam_back_node]: Time taken by CUVSLAM_WarmUpGPU(): 8.206451
[component_container-3] [INFO] [1729583960.004840366] [visual_slam_back.visual_slam_back_node]: VisualSlamNode end.
[INFO] [launch_ros.actions.load_composable_nodes]: Loaded node '/visual_slam_back/visual_slam_back_node' in container '/visual_slam_back_launch_container'

There is no similar question on issues. I am looking forward to you! Thanks.

silence-moon, 2024-10-22:
I call the callback function success. But there is no error, no observed cloud points and no positions.
[image]

silence-moon, 2024-10-22:
[component_container-5] Error:   TF_NAN_INPUT: Ignoring transform for child_frame_id "base_link" from authority "Authority undetectable" because of a nan value in the transform (nan nan nan) (nan -nan -nan nan)
[component_container-5]          at line 235 in /opt/ros/humble/src/geometry2/tf2/src/buffer_core.cpp
[component_container-5] Error:   TF_DENORMALIZED_QUATERNION: Ignoring transform for child_frame_id "base_link" from authority "Authority undetectable" because of an invalid quaternion in the transform (nan -nan -nan nan)
[component_container-5]          at line 254 in /opt/ros/humble/src/geometry2/tf2/src/buffer_core.cpp
[ekf_node-7] [ERROR] [1729588991.889939511] [ekf_filter_node]: Critical Error, NaNs were detected in the output state of the filter. This was likely due to poorly coniditioned process, noise, or sensor covariances.
[ekf_node-7] [ERROR] [1729588991.923438088] [ekf_filter_node]: Critical Error, NaNs were detected in the output state of the filter. This was likely due to poorly coniditioned process, noise, or sensor covariances.
[ekf_node-7] Error:   TF_NAN_INPUT: Ignoring transform for child_frame_id "base_link" from authority "Authority undetectable" because of a nan value in the transform (nan nan nan) (nan -nan -nan nan)
[ekf_node-7]          at line 235 in /opt/ros/humble/src/geometry2/tf2/src/buffer_core.cpp
[ekf_node-7] Error:   TF_DENORMALIZED_QUATERNION: Ignoring transform for child_frame_id "base_link" from authority "Authority undetectable" because of an invalid quaternion in the transform (nan -nan -nan nan)
[ekf_node-7]          at line 254 in /opt/ros/humble/src/geometry2/tf2/src/buffer_core.cpp
[component_container-2] Error:   TF_NAN_INPUT: Ignoring transform for child_frame_id "base_link" from authority "Authority undetectable" because of a nan value in the transform (nan nan nan) (nan -nan -nan nan)

closed 2024-11-08 by silence-moon

REGATTE, 2025-01-09:
How did you fix this??