NVIDIA-ISAAC-ROS/isaac_ros_visual_slam #191: Failed to initialize CUVSLAM tracker: 4

siddhu003, 2025-01-31:
I setup the Isaac ros visual slam packages on  x86_64 but while I'm running this command
`ros2 launch isaac_ros_examples isaac_ros_examples.launch.py \
launch_fragments:=zed_stereo_rect,visual_slam pub_frame_rate:=30.0 \
base_frame:=zed2_camera_center camera_optical_frames:="['zed2_left_camera_optical_frame', 'zed2_right_camera_optical_frame']" \
interface_specs_file:=${ISAAC_ROS_WS}/isaac_ros_assets/isaac_ros_visual_slam/zed2_quickstart_interface_specs.json`

[image]

I'm using x86_64, Ubuntu 22.04 and Zed 2 Camera.

Can someone please help me with this??

MeghaMaheshwari, 2025-03-28:
I m having the same issue. When i try with the examples provided also I get the same issue

[component_container-1] [INFO] [1743119732.966404516] [visual_slam_node]: Use use_gpu: true
[component_container-1] [INFO] [1743119732.966422707] [visual_slam_node]: Enable IMU Fusion: false
[component_container-1] [INFO] [1743119732.966436986] [visual_slam_node]: Time taken by CUVSLAM_CreateTracker(): 0.000001
[component_container-1] [ERROR] [1743119732.966452124] [visual_slam_node]: Failed to initialize CUVSLAM tracker: 4

I am using x86_64, Ubuntu 24.04 and Issac sim

zhangpeng66, 2025-05-23:
> I m having the same issue. When i try with the examples provided also I get the same issue
>
> [component_container-1] [INFO] [1743119732.966404516] [visual_slam_node]: Use use_gpu: true [...]

me too

zwdoescode, 2025-05-27:
Hi all! We updated the [verbosity](https://github.com/NVIDIA-ISAAC-ROS/isaac_ros_visual_slam/blob/main/isaac_ros_visual_slam/src/visual_slam_node.cpp#L106) level to 1 (ERROR) recently. Please try to pull the latest change and test it again. You should get more information about the error. 

Also please check GPU driver and CUDA version to see if they are compatible.